armink/FlashDB #231: install Flash DB on ESP32

Yukselzngn, 2023-08-14:
Hi Armink,
Can you please give me details how to install Flash DB on ESP32

armink, 2023-08-14:
you can see ESP demo https://github.com/armink/FlashDB/tree/master/demos/esp32_spi_flash

Yukselzngn, 2023-08-15:
Thank you for your answer.
I applied the database as in the example, but it gives this error while building.
../components/dbsrc/fal_flash_esp32_port.c:8:10: fatal error: esp_partition.h: No such file or directory
     8 | #include <esp_partition.h>
       | ^~~~~~~~~~~~~~~~~
compilation terminated.

When I check the file exists and is defined but it gives this error.

Yukselzngn, 2023-08-15:
FlashDB ESP32 SPI Flash Demo
This is ESP32 chip with 2 CPU cores, WiFi, silicon revision 300, [I/FAL] Flash Abstraction Layer (V0.5.99) initialize success.

assert failed: esp_partition_read partition_target.c:29 (partition != NULL)

ı have this error

closed 2023-08-16 by Yukselzngn