armink/FlashDB #235: 读写地址异常问题

MaRuolang, 2023-08-29:
芯片data flash只支持以uint32_t数据类型进行读写,而且要求地址是4的倍数,运行测试程序发现有时候读写地址不是4的倍数,会导致写入失败,请问怎么解决.FDB_WRITE_GRAN宏定义已配置为32.

armink, 2023-08-29:
要看下具体问题的函数调用栈，才好分析呢

MaRuolang, 2023-08-30:
我的描述可能不够清晰, 具体情况如下:
fal port 的 static int write(long offset, const uint8_t *buf, size_t size) 被调用时, 传入的 offset 可能不是 4 的倍数, 但是芯片官方的 BSP 提供的读写函数, 操作地址必须是 4 的倍数, 我该如何处理.
芯片官方读写函数如下:
/**
  * @brief Execute ISP FMC_ISPCMD_PROGRAM to program a word to flash.
  * @param[in]  u32Addr Address of the flash location to be programmed.
  *             It must be a word aligned address.
  * @param[in]  u32Data The word data to be programmed.
  * @return   0   Success
  * @return   -1  Program Failed
  *
  * @note     Global error code g_FMC_i32ErrCode
  *           -1  Program failed or time-out
  */
int32_t FMC_Write(uint32_t u32Addr, uint32_t u32Data)
{
    int32_t  tout;

    g_FMC_i32ErrCode = 0;
    FMC->ISPCMD = FMC_ISPCMD_PROGRAM;
    FMC->ISPADDR = u32Addr;
    FMC->ISPDAT = u32Data;
    FMC->ISPTRG = FMC_ISPTRG_ISPGO_Msk;

    tout = FMC_TIMEOUT_WRITE;
    while ((tout-- > 0) && (FMC->MPSTS & FMC_MPSTS_MPBUSY_Msk)) {}
    if (tout <= 0)
    {
        g_FMC_i32ErrCode = -1;
        return -1;
    }

    if (FMC->ISPSTS & FMC_ISPSTS_ISPFF_Msk)
    {
        FMC->ISPSTS |= FMC_ISPSTS_ISPFF_Msk;
        g_FMC_i32ErrCode = -1;
        return -1;
    }
    return 0;
}

armink, 2023-08-30:
我的意思是，你在是在什么调用栈上下文发现的这个问题，如何复现的？

JuneWng, 2023-08-31:
我也有类似问题，调试经常出现Error: The KV @0x000025A4 length has an error，使用示例代码测试，有时候正确，有时候错误，并且每次错误问题不太一样，本flash芯片sector大小为128字节，write_gran为32bit，麻烦作者能够指出问题所在，谢谢

MaRuolang, 2023-09-04:
我的芯片 dataflash 页大小是 4096.
参考 stm32f103ve 例程修改进行测试的, 是通过添加以下打印代码发现写入的地址不是 4 的倍数.  

```c
static int write(long offset, const uint8_t *buf, size_t size)
{
    if (offset % 4)
    {
        log_e("write offset error, offset = %d, size = %d", offset, size);
    }
    // 以下省略
}
```

我现在修改了写入函数, 让他把没写的字节补充 0xFF (日志 1), 代码如下:  

```c
static int write(long offset, const uint8_t *buf, size_t size)
{
    // log_i("write offset = %d, size = %d", offset, size);

    if (size == 0)
    {
        return 0;
    }

    if (offset % 4)
    {
        log_e("write offset error, offset = %d, size = %d", offset, size);
    }

    if (size % 4)
    {
        log_e("write size error, offset = %d, size = %d", offset, size);
    }

    uint32_t addr = m483_onchip_flash.addr + offset;
    uint32_t remain_addr = (addr + 3) / 4 * 4 - addr;
    uint32_t len = size;

    __attribute__((aligned(4))) uint32_t write_data;
    __attribute__((aligned(4))) uint32_t read_data;

    SYS_UnlockReg();
    FMC_Open();

    if (remain_addr != 0)
    {
        int i;
        uint32_t addr_temp = addr / 4 * 4;
        uint8_t *p = (uint8_t *)&write_data;
        for (i = 0; i < 4 - remain_addr; i++)
        {
            p[i] = 0xFF;
        }
        for (; i < 4; i++, buf++)
        {
            p[i] = *buf;
        }
        FMC_Write(addr_temp, write_data);
        read_data = *(__IO uint32_t *)addr_temp;
        if (_compare(write_data, read_data))
        {
            log_e("write error, addr = 0x%x, read_data = 0x%x != write_data = 0x%x", addr_temp, read_data, write_data);
        }

        addr += remain_addr;
        len -= remain_addr;
        _feed_dog();
    }
[... 1146 lines cut ...]
D/HEX rx_buf: 0520-053F: FF FF FF 4B 56 30 30 38  00 00 00 3D 8C D0 28 04  FF FF FF 03 00 00 00 FF  FF FF 74 65 6D 70 FF FF    ...KV008...=..(...........temp..
D/HEX rx_buf: 0540-055F: FF 33 38 43 FF FF FF 00  FF FF FF 00 FF FF FF 00  FF FF FF 00 FF FF FF FF  FF FF FF 4B 56 30 30 39    .38C.......................KV009
D/HEX rx_buf: 0560-057F: 00 00 00 11 95 68 CD 04  FF FF FF 04 00 00 00 FF  FF FF 74 65 6D 70 FF FF  FF 24 00 00 00 FF FF FF    .....h............temp...$......
D/HEX rx_buf: 0580-059F: 00 FF FF FF 00 FF FF FF  FF FF FF FF 00 FF FF FF  FF FF FF FF 4B 56 30 30  39 00 00 00 38 84 F7 5A    ....................KV009...8..Z
D/HEX rx_buf: 05A0-05BF: 04 FF FF FF 04 00 00 00  FF FF FF 74 65 6D 70 FF  FF FF 26 00 00 00 FF FF  FF FF FF FF FF FF FF FF    ...........temp...&.............
D/HEX rx_buf: 05C0-05DF: FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF    ................................
D/HEX rx_buf: 05E0-05FF: FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF    ................................
D/HEX rx_buf: 0600-061F: FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF    ................................
D/HEX rx_buf: 0620-063F: FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF    ................................
D/HEX rx_buf: 0640-065F: FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF    ................................
D/HEX rx_buf: 0660-067F: FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF    ................................
D/HEX rx_buf: 0680-069F: FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF    ................................
D/HEX rx_buf: 06A0-06BF: FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF    ................................
D/HEX rx_buf: 06C0-06DF: FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF    ................................
D/HEX rx_buf: 06E0-06FF: FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF    ................................
D/HEX rx_buf: 0700-071F: FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF    ................................
D/HEX rx_buf: 0720-073F: FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF    ................................
D/HEX rx_buf: 0740-075F: FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF    ................................
D/HEX rx_buf: 0760-077F: FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF    ................................
D/HEX rx_buf: 0780-079F: FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF    ................................
D/HEX rx_buf: 07A0-07BF: FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF    ................................
D/HEX rx_buf: 07C0-07DF: FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF    ................................
D/HEX rx_buf: 07E0-07FF: FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF    ................................
letter:/$
```

qianchenzhumeng, 2023-09-27:
我在 STM32L4xx 上也遇到了该问题，原因是 `_fdb_write_status` 函数内部调用 `_fdb_flash_write` 函数时的传参有问题：

```
#if (FDB_WRITE_GRAN == 1)
    result = _fdb_flash_write(db, addr + byte_index, (uint32_t *)&status_table[byte_index], 1, sync);
#else /*  (FDB_WRITE_GRAN == 8) ||  (FDB_WRITE_GRAN == 32) ||  (FDB_WRITE_GRAN == 64) */
    /* write the status by write granularity
     * some flash (like stm32 onchip) NOT supported repeated write before erase */
    result = _fdb_flash_write(db, addr + byte_index, (uint32_t *) &status_table[byte_index], FDB_WRITE_GRAN / 8, sync);
#endif /* FDB_WRITE_GRAN == 1 */
```

`FDB_WRITE_GRAN` 大于 1 时，写入的字节数比 `status_table` 的字节数要多，会覆盖后面的内容。如 `_fdb_set_status` 函数内部的说明，不同的 `FDB_WRITE_GRAN` 下，`status_table` 的长度不同：

```
    /*
     * | write garn |       status0       |       status1       |      status2         |       status3      |
     * ------------------------------------------------------------------------------------------------------
     * |    1bit    | 0xFF                | 0x7F                |  0x3F                |  0x1F
     * ------------------------------------------------------------------------------------------------------
     * |    8bit    | 0xFF FF FF          | 0x00 FF FF          |  0x00 00 FF          |  0x00 00 00
     * ------------------------------------------------------------------------------------------------------
     * |   32bit    | 0xFFFFFFFF FFFFFFFF | 0x00FFFFFF FFFFFFFF |  0x00FFFFFF 00FFFFFF |  0x00FFFFFF 00FFFFFF
     * |            | 0xFFFFFFFF          | 0xFFFFFFFF          |  0xFFFFFFFF          |  0x00FFFFFF
     * ------------------------------------------------------------------------------------------------------
     * |            | 0xFFFFFFFF FFFFFFFF | 0x00FFFFFF FFFFFFFF |  0x00FFFFFF FFFFFFFF |  0x00FFFFFF FFFFFFFF
     * |   64bit    | 0xFFFFFFFF FFFFFFFF | 0xFFFFFFFF FFFFFFFF |  0x00FFFFFF FFFFFFFF |  0x00FFFFFF FFFFFFFF
     * |            | 0xFFFFFFFF FFFFFFFF | 0xFFFFFFFF FFFFFFFF |  0xFFFFFFFF FFFFFFFF |  0x00FFFFFF FFFFFFFF
     */
```

但是 `_fdb_write_status` 传给 `_fdb_flash_write`  的参数 `size` 的值很奇怪，`FDB_WRITE_GRAN` 是 32 时传入的是 4，`FDB_WRITE_GRAN` 是 64 时，传入的是 8。按照 `port\fal\samples\porting\fal_flash_stm32f1_port.c` 和 `port\fal\samples\porting\fal_flash_stm32l4.c` 的示例代码分析，实际写入的字节分别是 32 字节和 64 字节，远远超过了 `status_table` 的大小，导致 flash 中 `kvdb_sec_info` 对应的信息损坏，包括 `magic` 在内的很多字段被改写了，导致再次初始化时触发格式化。

看了多处调用 `_fdb_flash_write` 的代码，`size` 的单位应该是字节，但是 `_fdb_write_status` 调用 `_fdb_flash_write` 时传入的 `size` 单位不是，`FDB_WRITE_GRAN` 大于 1 时，单位是 `FDB_WRITE_GRAN/8` 字节。然而， port 文件中，stm32f1 的代码视 `size` 的单位为 4 字节，而 stm32l4 的代码视 `size` 的单位为 8 字节，所以不管是站在字节的角度还是 `FDB_WRITE_GRAN / 8` 字节的角度，示例驱动和 FlashDB 的源码都无法配合起来。

qianchenzhumeng, 2023-09-27:
> 我在 STM32L4xx 上也遇到了该问题，原因是 `_fdb_write_status` 函数内部调用 `_fdb_flash_write` 函数时的传参有问题：
>
> ```
> #if (FDB_WRITE_GRAN == 1)
>     result = _fdb_flash_write(db, addr + byte_index, (uint32_t [...]

抱歉，我理解错了，`_fdb_write_status` 调用 `_fdb_flash_write` 时，`size` 的单位也是字节。`status_table` 中有两张表，每张表 4 个状态，每个状态 8 字节，更新状态时写入的字节确实是 8 字节。问题出在示例适配驱动上面，如之前的内容所述，示例适配驱动程序中 `write` 函数的传参 `size` 的单位是 `FDB_WRITE_GRAN/ 8` 字节。

修改方案：修改适配驱动，将 `write` 函数的传参 `size` 的单位按字节处理。

另外就是注意片内 Flash 擦写的问题，FlashDB 貌似不像 FatFS 那样写之前会回读，需要自己实现回读机制。

closed 2023-11-23 by MaRuolang

y30190753, 2025-05-05:
我这边也遇到了地址相关的错误：
[FlashDB][sample][kvdb][basic] ==================== kvdb_basic_sample ====================
[FlashDB][sample][kvdb][basic] first set the 'boot_count' value to 0
[FlashDB][kv][env][easyflash] Error: Read the KV (
@0x00000010) CRC32 check failed!
[FlashDB][kv][env][easyflash] Error: The KV @0x00000054 length has an error.
[FlashDB][sample][kvdb][basic] get the 'boot_count' value is 0
[FlashDB][kv][env][easyflash] Error: Read the KV (
@0x00000010) CRC32 check failed!
[FlashDB][kv][env][easyflash] Error:

[2025-05-05 20:07:50.456]# RECV ASCII>
 Read the KV (
@0x00000010) CRC32 check failed!
[FlashDB][kv][env][easyflash] Error: The KV @0x00000054 length has an error.
[FlashDB][sample][kvdb][basic] set the 'boot_count' value to 1
[FlashDB][kv][env][easyflash] Error: Read the KV (
@0x00000010) CRC32 check failed!
[FlashDB][kv][env][easyflash] Error: The KV @0x00000054 length has an error.
[FlashDB][sample][kvdb][basic] get the 'boot_count' value is 0
[FlashDB][sample][kvdb][basic] ===========================================================

是将FlashDBv.2.1.1移植到stm32h743iit6上，nor flash使用w25q64jv时遇到的，还不清楚怎么调整驱动。
再指教一下呀~~~~~