armink/FlashDB #242: tsdb 在空间耗尽  擦除旧数据写入新数据后的两个bug

cld0815, 2023-09-26:
使用的提交: master 最新 (3e441f6c26590eb86051826830b19a715d3d7423)

1. 调用 fdb_tsdb_init 后,得到的 last_time 值为 0
[image]
[image]

2. 调用fdb_tsl_iter_by_time, from < to, to = INT32_MAX, 返回的记录中, 没有从最小的(from)开始取
举例: from = 2000   to = INT32_MAX
取回的记录的 time: 2010, 2000, 2001, .....

上述问题均为flash空间全部写了一遍或更多后, 才出现

谢谢~

armink, 2023-10-06:
看着似乎和起始扇区的查找有关系，可否深入分析下问题呢？

bu-butterfly, 2025-03-13:
[image]
我遇到了类似的问题，记录不断保存直到超过分配的空间上限，flashDB自动回滚到addr为0的位置再存储，我再用fdb_tsl_query_count去查找这段时间的记录数量时明显偏少；
我看了一下update_sec_status函数的实现，感觉是flash满后强制将db_oldest_addr置为0了，导致只查询到（0~最近记录addr）的记录；
能否增加以下需求：判断要清空的扇区后面的扇区是否有数据，以此为依据将db_oldest_addr分情况赋值