armink/FlashDB #259: 关于默认参数表写入超过48字节的字符型kv引发的问题

BurnerK, 2023-11-15:
先是在默认参数表中添加了`    {EB_PARAMINDEX_EN_MQTT_CUSTPUB,"OFF,OFF,OFF,OFF,OFF,OFF,OFF,OFF,OFF,OFF,OFF,OFF,OFF,OFF,OFF,OFF,", 0},
`，该默认值长度为63个字符，根据日志显示，参数库正常初始化无报错，但在读取该参数时，会在`int fal_partition_read(const struct fal_partition *part, uint32_t addr, uint8_t *buf, size_t size) {
    int                         ret       = 0;
    const struct fal_flash_dev *flash_dev = NULL;

    assert(part);
    assert(buf);

    if (addr + size > part->len) {
        log_e("Partition read error! Partition address out of bound.");
        return -1;
    }

    flash_dev = flash_device_find_by_part(part);
    if (flash_dev == NULL) {
        log_e("Partition read error! Don't found flash device(%s) of the partition(%s).", part->flash_name, part->name);
        return -1;
    }

    ret = flash_dev->ops.read(part->offset + addr, buf, size);
    if (ret < 0) {
        log_e("Partition read error! Flash device(%s) read error!", part->flash_name);
    }

    return ret;
}`中的` if (addr + size > part->len) {
        log_e("Partition read error! Partition address out of bound.");
        return -1;
    }`显示出错，添加打印信息，发现`addr`是一个大的不正常的值。
请问可能导致发生该错误的原因有哪些，如何排查？

songchong11, 2023-12-09:
我也遇到了这个问题，跑 KVDB Sample 的时候回遇到。还没解决。

[FlashDB][kv][env][fdb_kvdb1] (../vendor/flash_db/src/fdb_kvdb.c:1739) The oldest addr is @0x00007000 /*PRIX32*/ 
[FlashDB][kv][env][fdb_kvdb1] (../vendor/flash_db/src/fdb_kvdb.c:1753) KVDB size is 8000/*PRIu32*/  bytes.
[31;22m[E/FAL] (fal_partition_read:419) Partition read error! Partition address out of bound. addr 7fe6, size 20, part->len 8000[0m
[31;22m[E/FAL] (fal_partition_read:419) Partition read error! Partition address out of bound. addr 7fe6, size 20, part->len 8000[0m
[31;22m[E/FAL] (fal_partition_read:419) Partition read error! Partition address out of bound. addr 7fe6, size 20, part->len 8000[0m
[31;22m[E/FAL] (fal_partition_read:419) Partition read error! Partition address out of bound. addr 7fe6, size 20, part->len 8000[0m
[31;22m[E/FAL] (fal_partition_read:419) Partition read error! Partition address out of bound. addr 7fe6, size 20, part->len 8000[0m
[31;22m[E/FAL] (fal_partition_read:419) Partition read error! Partition address out of bound. addr 7fe6, size 20, part->len 8000[0m
[31;22m[E/FAL] (fal_partition_read:419) Partition read error! Partition address out of bound. addr 7fe6, size 20, part->len 8000[0m
[FlashDB][sample][kvdb][basic] ==================== kvdb_basic_sample ====================
[FlashDB][sample][kvdb][basic] get the 'boot_count' value is 344
[FlashDB][sample][kvdb][basic] set the 'boot_count' value to 345
[FlashDB][sample][kvdb][basic] ===========================================================
[FlashDB][sample][kvdb][string] ==================== kvdb_type_string_sample ====================
[31;22m[E/FAL] (fal_partition_read:419) Partition read error! Partition address out of bound. addr 7fe6, size 20, part->len 8000[0m
[FlashDB][sample][kvdb][string] create the 'temp' string KV, value is: 36C
[FlashDB][sample][kvdb][string] get the 'temp' value is: 36C
[FlashDB][sample][kvdb][string] set 'temp' value to 38C
[FlashDB][sample][kvdb][string] delete the 'tem?? finish
[FlashDB][sample][kvdb][string] ===========================================================
[FlashDB][sample][kvdb][blob] ==================== kvdb_type_blob_sample ====================
[31;22m[E/FAL] (fal_partition_read:419) Partiti?鹡 read error! Partition address out of bound. addr 7fe6, size 20, part->len 8000[0m
[FlashDB][sample][kvdb][blob] create the 'temp' blob KV, value is: 36
[FlashDB][sample][kvdb][blob] get the 'temp' value is: 36
[FlashDB][sample][kvdb][blob] set 'temp' value to 38
[FlashDB][sample][kvdb][blob] delete the 'temp' finish

BurnerK, 2023-12-09:
> 我也遇到了这个问题，跑 KVDB Sample 的时候回遇到。还没解决。
>
> [FlashDB][kv][env][fdb_kvdb1] (../vendor/flash_db/src/fdb_kvdb.c:1739) The oldest addr is @0x00007000 /_PRIX32_/ [FlashDB][kv][env][fdb_kvdb1] [...]

我发现用创建blob参数的方式创建就没问题了，定义一个数组，用来存默认值长度比较大的参数。
```
static char sg_param_mqtt_pub_enabled[65] = "OFF,OFF,OFF,OFF,OFF,OFF,OFF,OFF,OFF,OFF,OFF,OFF,OFF,OFF,OFF,OFF";

static struct fdb_default_kv_node sg_main_flash_param_table[]={
    {"param", &sg_param_mqtt_pub_enabled, sizeof(sg_param_mqtt_pub_enabled)},
}
```
- [ ]

songchong11, 2023-12-09:
您好，你的邮件已收到。

closed 2023-12-15 by BurnerK