armink/FlashDB #275: 测试tsdb连续保存累计数达一定数量后出现前面部分数据丢失的现象

biqi2, 2024-01-10:
1、用的例程为‘stm32f103ve’，STM32单片机型号为STM32F105RBT6，其FLASH=128Kbyte，RAM=64Kbyte，其blk_size=2K，相关配置如下
[image]
[image]
[image]
[image]
从上面可见tsdb的缓存范围为对应SMT32内部FLASH的第50K地址处开始直到54K-1地址处结束.
2、在main函数中两个例子/* KVDB Sample */、/* TSDB Sample */之后增加了while循环反复执行这两个例子中的核心函数kvdb_basic_sample(&kvdb)及tsdb_sample(&tsdb);main代码如下：
int temp_int;
int main(void)
{
    fdb_err_t result;
	
//	print_stm32_ic_information();//打印STM32芯片信息	
#ifdef FDB_USING_KVDB
    { /* KVDB Sample */
        struct fdb_default_kv default_kv;

        default_kv.kvs = default_kv_table;
        default_kv.num = sizeof(default_kv_table) / sizeof(default_kv_table[0]);
        /* set the lock and unlock function if you want */
        fdb_kvdb_control(&kvdb, FDB_KVDB_CTRL_SET_LOCK, (void *)lock);
        fdb_kvdb_control(&kvdb, FDB_KVDB_CTRL_SET_UNLOCK, (void *)unlock);
        /* Key-Value database initialization
         *
         *       &kvdb: database object
         *       "env": database name
         * "fdb_kvdb1": The flash partition name base on FAL. Please make sure it's in FAL partition table.
         *              Please change to YOUR partition name.
         * &default_kv: The default KV nodes. It will auto add to KVDB when first initialize successfully.
         *        NULL: The user data if you need, now is empty.
         */
        result = fdb_kvdb_init(&kvdb, "env", "fdb_kvdb1", &default_kv, NULL);

        if (result != FDB_NO_ERR) {
            return -1;
        }

        /* run basic KV samples */
        kvdb_basic_sample(&kvdb);
        /* run string KV samples */
        kvdb_type_string_sample(&kvdb);
        /* run blob KV samples */
        kvdb_type_blob_sample(&kvdb);
    }
#endif /* FDB_USING_KVDB */

#ifdef FDB_USING_TSDB
    { /* TSDB Sample */
        /* set the lock and unlock function if you want */
        fdb_tsdb_control(&tsdb, FDB_TSDB_CTRL_SET_LOCK, (void *)lock);
        fdb_tsdb_control(&tsdb, FDB_TSDB_CTRL_SET_UNLOCK, (void *)unlock);
        /* Time series database initialization
         *
         *       &tsdb: database object
         *       "log": database name
         * "fdb_tsdb1": The flash partition name base on FAL. Please make sure it's in FAL partition table.
         *              Please change to YOUR partition name.
         *    get_time: The get current timestamp function.
         *         128: maximum length of each log
         *        NULL: The user data if you need, now is empty.
         */
        result = fdb_tsdb_init(&tsdb, "log", "fdb_tsdb1", get_time, 128, NULL);
        /* read last saved time for simulated timestamp */
        fdb_tsdb_control(&tsdb, FDB_TSDB_CTRL_GET_LAST_TIME, &counts);
		//这一步的目的是将tsdb对象中的last_time成员恢复到静态变量counts中；因为上一行fdb_tsdb_init（）函数
		//在上电后初始化ts类型的fdb数据库时调用函数get_time（）得到变量counts(初始值0累加1后的结果)=1作为当前时间戳,显然这个当前时间戳只有在下载程序后的首次运行时正确的
		//在进行了计秒操作后重新上电得到的这个是不正确的。因此就需要将tsdb对象中的last_time成员赋值给计秒变量。
		
		//在这个例程中静态变量counts的值代表的是当前模拟的时间戳值，时间戳本意是指格林威治时间1970年01月01日00时00分00秒（北京时间1970年01月01日08时00分00秒）起至现在的总秒数。
		//由于这个例程没有建立时间计秒机制，因此这里使用get_time()函数来模拟计秒动作（对counts累加一次），
[... 269 lines cut ...]
[FlashDB][sample][tsdb] [query_cb] queried a TSL: time: 85, temp: 85, humi: 85
[FlashDB][sample][tsdb] [query_cb] queried a TSL: time: 86, temp: 86, humi: 86
[FlashDB][sample][tsdb] [query_cb] queried a TSL: time: 87, temp: 87, humi: 87
[FlashDB][sample][tsdb] [query_cb] queried a TSL: time: 88, temp: 88, humi: 88
[FlashDB][sample][tsdb] [query_cb] queried a TSL: time: 89, temp: 89, humi: 89
[FlashDB][sample][tsdb] [query_cb] queried a TSL: time: 90, temp: 90, humi: 90
[FlashDB][sample][tsdb] [query_cb] queried a TSL: time: 91, temp: 91, humi: 91
[FlashDB][sample][tsdb] [query_cb] queried a TSL: time: 92, temp: 92, humi: 92
[FlashDB][sample][tsdb] [query_cb] queried a TSL: time: 93, temp: 93, humi: 93
[FlashDB][sample][tsdb] [query_cb] queried a TSL: time: 94, temp: 94, humi: 94
[FlashDB][sample][tsdb] [query_cb] queried a TSL: time: 95, temp: 95, humi: 95
[FlashDB][sample][tsdb] [query_cb] queried a TSL: time: 96, temp: 96, humi: 96
[FlashDB][sample][tsdb] [query_cb] queried a TSL: time: 97, temp: 97, humi: 97
[FlashDB][sample][tsdb] [query_cb] queried a TSL: time: 98, temp: 98, humi: 98
[FlashDB][sample][tsdb] [query_cb] queried a TSL: time: 99, temp: 99, humi: 99
[FlashDB][sample][tsdb] ===========================================================
[FlashDB][sample][kvdb][basic] ==================== kvdb_basic_sample ====================
[FlashDB][sample][kvdb][basic] get the 'boot_count' value is 99
[FlashDB][sample][kvdb][basic] set the 'boot_count' value to 100
[FlashDB][sample][kvdb][basic] ===========================================================
[FlashDB][sample][tsdb] ==================== tsdb_sample ====================
[FlashDB][sample][tsdb] append the new status.temp (100) and status.humi (100)
[FlashDB][sample][tsdb] [query_cb] queried a TSL: time: 50, temp: 50, humi: 50
[FlashDB][sample][tsdb] [query_cb] queried a TSL: time: 51, temp: 51, humi: 51
[FlashDB][sample][tsdb] [query_cb] queried a TSL: time: 52, temp: 52, humi: 52

biqi2, 2024-01-12:
用API函数fdb_tsdb_control禁止ts数据库覆盖后发现在4K字节的TS分区的情况下确实只能容纳98条记录，因此重新分析fdb数据记录格式，发现：
  每个扇区的扇区头结构体（sector_hdr_data）大小为80字节，每条数据的数据头结构体（log_idx_data）大小为32字节，因此 整个TSdb分区可以储存TS记录的数量 = (分区大小 - 扇区数*80) / （32 + 每条记录的数据对应的大小 ），由这个公式可计算对应大小TSdb分区所能包含的记录的条数，于是有了下面我的这个记录表格。由表格记录可见随着TSdb分区的增大，每多增加10K字节就会发生TS记录减少1条的情况（我的每条记录数据大小为8字节），推测可能TSdb分区每多10K产生了额外的空间占用（没有详细去分析源码了），或者我上面的公式不够精确？
[image]

turelon, 2024-05-16:
> 用API函数fdb_tsdb_control禁止ts数据库覆盖后发现在4K字节的TS分区的情况下确实只能容纳98条记录，因此重新分析fdb数据记录格式，发现： 每个扇区的扇区头结构体（sector_hdr_data）大小为80字节，每条数据的数据头结构体（log_idx_data）大小为32字节，因此 整个TSdb分区可以储存TS记录的数量 = (分区大小 - 扇区数*80) / （32 + [...]

你这表格计算有问题，记录条数应该取整，按照你的公式，每个扇区可以存储的条数为49.2，取整为49。否则0.2的小数累计下来就会多出来