armink/FlashDB #285: linux的demo程序第一次运行都会卡住

syan-cham, 2024-03-28:
FlashDB-2.1.0/demos/linux$ ./out/FlashDBLinuxDemo 
[FlashDB][kv][env][fdb_kvdb1] (../../src/fdb_kvdb.c:1777) The oldest addr is @0x00000000
[FlashDB][kv][env][fdb_kvdb1] (../../src/fdb_kvdb.c:1793) KVDB size is 16384 bytes.
[FlashDB][kv][env][fdb_kvdb1] Sector header info is incorrect. Auto format this sector (0x00000000).
[FlashDB][kv][env][fdb_kvdb1] Sector header info is incorrect. Auto format this sector (0x00001000).
[FlashDB][kv][env][fdb_kvdb1] Sector header info is incorrect. Auto format this sector (0x00002000).
[FlashDB][kv][env][fdb_kvdb1] Sector header info is incorrect. Auto format this sector (0x00003000).
[FlashDB][kv][env][fdb_kvdb1] All sector header is incorrect. Set it to default.

这个地方就一直卡住，要ctrl+c停止再运行才能往下工作。
FlashDB-2.1.0/demos/linux$ ./out/FlashDBLinuxDemo 
[FlashDB][kv][env][fdb_kvdb1] (../../src/fdb_kvdb.c:1777) The oldest addr is @0x00000000
[FlashDB][kv][env][fdb_kvdb1] (../../src/fdb_kvdb.c:1793) KVDB size is 16384 bytes.
[FlashDB] FlashDB V2.1.0 is initialize success.
[FlashDB] You can get the latest version on https://github.com/armink/FlashDB .
[FlashDB][sample][kvdb][basic] ==================== kvdb_basic_sample ====================
[FlashDB][sample][kvdb][basic] get the 'boot_count' failed
[FlashDB][sample][kvdb][basic] set the 'boot_count' value to 1
[FlashDB][sample][kvdb][basic] ===========================================================
[FlashDB][sample][kvdb][string] ==================== kvdb_type_string_sample ====================
[FlashDB][sample][kvdb][string] create the 'temp' string KV, value is: 36C
[FlashDB][sample][kvdb][string] get the 'temp' value is: 36C
[FlashDB][sample][kvdb][string] set 'temp' value to 38C
[FlashDB][sample][kvdb][string] delete the 'temp' finish
[FlashDB][sample][kvdb][string] ===========================================================
[FlashDB][sample][kvdb][blob] ==================== kvdb_type_blob_sample ====================
[FlashDB][sample][kvdb][blob] create the 'temp' blob KV, value is: 36
[FlashDB][sample][kvdb][blob] get the 'temp' value is: 36
[FlashDB][sample][kvdb][blob] set 'temp' value to 38
[FlashDB][sample][kvdb][blob] delete the 'temp' finish
[FlashDB][sample][kvdb][blob] ===========================================================
[FlashDB][tsl][log][fdb_tsdb1] Sector (0x00000000) header info is incorrect.
[FlashDB][tsl][log][fdb_tsdb1] All sector format finished.
[FlashDB][tsl][log][fdb_tsdb1] (../../src/fdb_tsdb.c:978) TSDB (log) oldest sectors is 0x00000000, current using sector is 0x00000000.   
[FlashDB][sample][tsdb] ==================== tsdb_sample ====================
[FlashDB][sample][tsdb] append the new status.temp (36) and status.humi (85)
[FlashDB][tsl][log][fdb_tsdb1] Warning: current timestamp (1711640796) is less than or equal to the last save timestamp (1711640796). This tsl will be dropped.
[FlashDB][sample][tsdb] append the new status.temp (38) and status.humi (90)
[FlashDB][sample][tsdb] [query_cb] queried a TSL: time: 1711640796, temp: 36, humi: 85
[FlashDB][sample][tsdb] query count is: 0
[FlashDB][sample][tsdb] set the TSL (time 1711640796) status from 2 to 3
[FlashDB][sample][tsdb] ===========================================================

mrchipset, 2024-05-10:
今天看到这个项目想试一下也遇到这个问题。看了下发现是因为线程锁造成的阻塞。

`fdb_kvdb_init`这个函数里获得了线程锁，还没解锁前调用`_fdb_kv_load`中函数`fdb_kv_set_default`也想要获得线程锁就卡住了。
[image]

修改下代码`fdb_kvdb.c`文件中的`fdb_kvdb_init`函数可以修复
```c
FDB_DEBUG("KVDB size is %" PRIu32 " bytes.\n", db_max_size(db));
db_unlock(db);

result = _fdb_kv_load(db);

db_lock(db);
```

armink, 2024-05-10:
@mrchipset 多谢啦，估计是锁没有支持递归能力

armink, 2025-10-29:
fixed in #295

closed 2025-10-29 by armink