armink/FlashDB #301: tsdb 写数据的时候突然重启设备，重启后设备 db->last_time 变成0开始

shun-dgut, 2024-07-02:


shun-dgut, 2024-07-02:
`[FlashDB][kv][ate_db][ate_db] (packages\FlashDB-v1.1.2\src\fdb_kvdb.c:1801) KVDB size is 24576 bytes.
00> [FlashDB][tsl][record_db][record_db] (packages\FlashDB-v1.1.2\src\fdb_tsdb.c:981) TSDB (record_db) oldest sectors is 0x00000000, current using sector is 0x0001F000.
00> 4========db->last_time 0`

shun-dgut, 2024-07-03:
一个线程使用fdb_tsl_append间隔10ms不断写入数据，在 tsl_append 里面打印 db->last_time 的值，这个时候使用reboot命令重启设备，上电打印  db->last_time 值，会出现  db->last_time 变化为0的情况。
大概是四次五次使用reboot后就出现。

armink, 2024-07-03:
有没有用最新版本测试过呢？

shun-dgut, 2024-07-03:
> 有没有用最新版本测试过呢？
> 
> 最新版本是 v2.1.0 我看你的是 v1.1.2

使用的最新版本测试的，文件夹名字没有更改

armink, 2024-07-03:
那你先试着单步调试，分析一下获取 last_time 的逻辑？

armink, 2024-07-03:
[image]

这里看着是有做掉电保护的，你分析下，为何 last_time 变成 0 了?

shun-dgut, 2024-07-03:
> ![image](https://private-user-images.githubusercontent.com/1734686/345368516-b255c7be-05e5-4cbe-95f1-678b9d68798e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3L [...]

fdb_tsdb_init 这一块的逻辑有一些文档说明吗，有没有可能是这一块的逻辑有问题

shun-dgut, 2024-07-15:
[image]
[image]
[image]
[image]

我发现重启前db->cur_sec.addr的地址也是没有问题的，但是重启后，db->cur_sec.end_time 就是0值了

shun-dgut, 2024-07-15:
@armink 朱工，你看一下是不是这里导致的，我提交了一个pr https://github.com/armink/FlashDB/pull/302