armink/FlashDB #310: esp32S3创建多个分区，两个kvdb,两个tsdb,每次上电，第二个kvdb总是重新格式化扇区

Alex-H-C, 2024-09-02:
`#ifdef FAL_PART_HAS_TABLE_CFG
/* partition table */
#define FAL_PART_TABLE                                                               \
{                                                                                    \
    {FAL_PART_MAGIC_WORD, "fdb_kvdb1", NOR_FLASH_DEV_NAME,            0,256*1024, 0},   \
    {FAL_PART_MAGIC_WORD, "log_kvdb", NOR_FLASH_DEV_NAME,256*1024,1024*1024, 0},  \
    {FAL_PART_MAGIC_WORD, "MCCB_tsdb1", NOR_FLASH_DEV_NAME, 1024*1024,2*1024*1024, 0}, \
    {FAL_PART_MAGIC_WORD, "MCB_tsdb2", NOR_FLASH_DEV_NAME,2*1024*1024, 6*1024*1024, 0}, \
}
#endif /* FAL_PART_HAS_TABLE_CFG */`
[image]
另外就是上电从kvdb中读取65*3=195个参数，时间太长，导致任务看门狗复位
[image]

Alex-H-C, 2024-09-02:
[image]
[image]

herexiong, 2024-09-08:
您好！请问你使用的是哪个版本的IDF，我使用IDF4.4.6运行示例时TSDB报错重启，KVDB正常。但是它们在IDF5.x版本却工作正常

penghongquan, 2024-09-09:
你没发现你分区有重合吗

Alex-H-C, 2024-09-09:
> 您好！请问你使用的是哪个版本的IDF，我使用IDF4.4.6运行示例时TSDB报错重启，KVDB正常。但是它们在IDF5.x版本却工作正常
我也用的4.4.6，tsdb报错，你看下是不是你数组访问越界了，一般是没有问题的