armink/FlashDB #312: FlashDB: Add Zephyr RTOS Support

karthi012, 2024-09-10:
Hi all. First of all thanks for this nice project.
Currently, I'm working on the integration of FlashDB with the Zephyr RTOS. From Zephyr's view,  FlashDB is added as external module, so that we consume in the application level. For now, FlashDB is working in POSIX file mode, in future, we can support for Flash mode.
In order to consume the FlashDB as external module we need to add a Zephyr's module in FlashDB project. 
Is it possible to add zephyr module directory in FlashDB project so that we avoid forks and no change is need in source of FlashDB.

Please take a look here https://github.com/armink/FlashDB/commit/c27d3b6f3ee43837c7ee8679876f54f6e77de642 for how to add
integration works for zephyr.
 
Is it possible to add zephyr module directory in FlashDB project?. Thanks in advance.

armink, 2024-09-10:
Yes, no problem

It is also amazing to as a external  module for Zephyr

karthi012, 2024-09-10:
@armink Thanks :).

Will open a PR.

Apologies for not mentioning previously, how modules consume in zephyr https://github.com/linumiz/rixx/commit/d54a84701b10436837c2febf9b8dd6a0bbeb54c6 (Note: this changes in this link is only for Zephyr, How the FlashDB is consumed).

karthi012, 2024-09-10:
Zephyr's RTOS Support PR https://github.com/armink/FlashDB/pull/313 got merged, So closing this issue. 

Thanks @armink.

closed 2024-09-10 by karthi012