armink/FlashDB #315: Error: The KV @0x00001024 length has an error.

BoTan-sudo, 2024-09-27:
[image]
你好！在使用KVDB时出现这个错误。并且在出现这个错误之后，单片机复位后，存储在KVDB中的参数会丢失。请问这个错误是什么原因导致的呢？谢谢~
[image]

weicao0, 2024-11-12:
我这边也遇到这个问题，是用的file模式。定位到，read_kv中获得的kv_hdr.len是-1。这个现象在我这边每隔一段时间就会出现。
请问你最终解决了吗？
[image]

BoTan-sudo, 2024-11-12:
我的问题解决了。我的问题的原因是 Flash 的擦写粒度( write_gran ) 和 read, write 函数里面操作参数不一致造成的。操作流程适配好 write_gran，我的问题就解决了。

 

 image.png (view on web)

BoTan-sudo, 2024-11-12:
[image]

BoTan-sudo, 2024-11-12:
Flash 的 page_size 也得设置好，这影响到擦除时的容量。
[image]

weicao0, 2024-11-12:
好吧。我的问题是FILE模式下的cache处理有问题，把FDB_KV_CACHE_TABLE_SIZE禁掉反而好了。

armink, 2024-11-15:
> 好吧。我的问题是FILE模式下的cache处理有问题，把FDB_KV_CACHE_TABLE_SIZE禁掉反而好了。

我也每天都在用 file 模式，还没有遇到你的说的问题，大概是咋样的使用场景？

weicao0, 2025-05-26:
[log.txt](https://github.com/user-attachments/files/20436057/log.txt)