armink/FlashDB #318: 请问，TSDB可以多进程使用吗，一写多读的场景

Pluto-kk, 2024-10-12:
测试，两个进程先后初始化tsdb数据库，进程1写入的数据，进程2读取不到数据

littlewhitezkz, 2024-10-16:
我觉得是可以的吧，只要你能够实现这些操作都是异步执行的。

armink, 2024-10-16:
应该不能两个进程同时打开数据库的，A 打开，操作，关闭后，B 才可以重新打开，再操作

littlewhitezkz, 2024-10-16:
> 应该不能两个进程同时打开数据库的，A 打开，操作，关闭后，B 才可以重新打开，再操作
hi，朱工，我想问下难道数据库A和B的是共用一个锁的吗？

在fdb_tsdb.c中db_lock(db),db_unlock(db);根据要输入db这一点来看，我理解为，
A和B个各具备一个锁，分别是由db_lock(A),db_unlock(B)开启
所以在两个进程中，理论上A可以进行操作，B也可以进行操作
but，您的描述是说B能够解锁要取决于A是否关闭
那难道说同一个时刻只能有一个数据库被操作，那这样一来何必有参数db，
都统一写成db_lock不就可以了吗？

littlewhitezkz, 2024-10-16:
> 应该不能两个进程同时打开数据库的，A 打开，操作，关闭后，B 才可以重新打开，再操作

hi，朱工，我想问下难道数据库A和B的是共用一个锁的吗
在fdb_tsdb.c中db_lock(db),db_unlock(db);根据要输入db这一点来看，我理解为，
A和B个各具备一个锁，分别是由db_lock(A),db_unlock(B)开启
所以在两个进程中，理论上A可以进行操作，B也可以进行操作
but，您的描述是说B能够解锁要取决于A是否关闭
那难道说同一个时刻只能有一个数据库被操作，那这样一来何必有参数db，
都统一写成db_lock不就可以了吗？

armink, 2024-10-16:
我上面说的 A 和 B 指的是进程 A 和进程 B ，操作的是同一个数据库文件，要注意不要同时打开

Pluto-kk, 2024-10-23:
原先测试，多进程同时打开同个数据库，进程A写，进程B读，A写入数据， B只能读到数据库初始化前已有的数据，读不到文件A追加的数据

Pluto-kk, 2024-10-23:
原先测试，多进程同时打开同个数据库，进程A写，进程B读，
    A写入数据， B只能读到数据库初始化前已有的数据，读不到文件A追加的数据
    A删除数据，B完全读不到数据
目前使用了一段时间，B每次读取前重新初始化一次数据库，B可以读取到A写入/删除的数据。

想问一下，明确是不支持多进程一写一读场景吗，初始化数据库那一块我没完全看懂，所以不太确定

armink, 2024-10-23:
应该是做不到了，你这个属于数据库服务器的需求了，得有个 CS 架构来实现

还有个主意，增加文件锁，每次 A 进程打开数据库，写入完后就关闭数据库，B 进程打开完数据库，读取完也关闭数据库

Pluto-kk, 2024-10-23:
好的，谢谢；