armink/FlashDB #334: dsp移植flashdb遇到的问题。

Ou-Fei, 2024-11-19:
因为dsp c2000系列的特殊，每个地址存储16位数据，不是像传统ARM那样8位存储，在移植过程中flash用的是w25q64，导致在写入数据时会出现错误的情况，请问这种情况如何解决适配。

armink, 2024-11-26:
如果不是 xip 方式访问 w25q64 ，通过 spi API 接口操作 w25q64 是不是就行

Ou-Fei, 2024-11-30:
意思是平常的arm芯片一个32位的变量，你把它写入flash，一般会转换成uint8_t 类型指针，spi写函数会在内部将指针++，分4次写入，每次1个字节，但是TI 的dsp芯片比较特殊，是每个地址存16位数据，最小数据单位16位也就是2个字节，就导致spi写函数将数据指针++就会直接跳过16位，而不是8位。ARM sizeof返回32位变量是4，TI DSP sizeof返回32位变量是2而不是4，就会导致flashdb在写入一些数据出现错误。

figght, 2025-05-21:
> 意思是平常的arm芯片一个32位的变量，你把它写入flash，一般会转换成uint8_t 类型指针，spi写函数会在内部将指针++，分4次写入，每次1个字节，但是TI 的dsp芯片比较特殊，是每个地址存16位数据，最小数据单位16位也就是2个字节，就导致spi写函数将数据指针++就会直接跳过16位，而不是8位。ARM sizeof返回32位变量是4，TI DSP sizeof返回32位变量是2而不是4，就会导致flashdb在写入一些数据出现错误。

请问解决了吗?