armink/FlashDB #335: GC的时机问题

alexyuoouy, 2024-11-21:
大家好，最近使用FlashDB时，遇到一个问题，由于在申请空间和写入完之后来判断是否需要进行一下GC，所以在程序里的任何调用fdb_kv_set_blob或者fdb_kv_set的地方，都有可能进行一次GC，这就要求所有使用KVDB的任务，都要留够一次GC所用到的栈空间，这对于资源较紧张的单片机来说不大友好，而且这样会使这个过程不可控，不知道它会什么时候运行，由于flash操作缓慢，它如果在高优先级任务中执行，那么还可能会影响到实时性。

我目前应用里是把所有自动GC都屏蔽掉了，然后又写了一个函数，放在一个周期任务中手动检查扇区剩余空间和剩余扇区来判断是否需要GC，这样会牺牲一些flash空间，因为手动调用没办法卡的那么合适，但是省了不少任务栈。

大家觉得这个问题是否是个问题？这么解决是否合适？

armink, 2024-11-26:
最新版本的 GC 并不会全盘进行垃圾回收，基本只会回收 1 个扇区。只是一个扇区的擦写，是不是影响不大了呢？

BurnerK, 2024-11-26:
> 最新版本的 GC 并不会全盘进行垃圾回收，基本只会回收 1 个扇区。只是一个扇区的擦写，是不是影响不大了呢？

请问GC时，大概会占用多少任务内存空间？

cwq1142173025, 2024-12-10:
我也遇到了跟楼主一样的问题, 触发gc时 会占用大量的栈空间, 这会导致我只要调用到 set 的任务都需要预留很大栈空间,  这样非常浪费

alexyuoouy, 2025-09-01:
> 最新版本的 GC 并不会全盘进行垃圾回收，基本只会回收 1 个扇区。只是一个扇区的擦写，是不是影响不大了呢？

您好朱工，有两个问题：1.没看到哪里控制的每次只回收一个扇区，我看代码里还是[do_gc](https://github.com/armink/FlashDB/blob/master/src/fdb_kvdb.c#L1109)的时候有个迭代器，然后每个扇区去判断dirty true或者dirty gc就会进去搬kv；2.是否可以将GC的过程可以有选择的放到一个单独的任务中，比如说GC可以放到工作队列里，只要工作队列的栈够用就行，有需要GC时去通知运行工作队列，对于裸机的话，就一个栈，就保持现在这样；这样使用者如果使用的裸机或者嫌麻烦做单独任务的，就直接用，不需要什么改动，而对于栈或者实时性有要求的，就将GC拿出来，需要自己再去按照给定的接口实现一些代码；总结就是，可以选择把GC执行的控制权交给用户，我们只需告诉用户要进行垃圾回收了，这个提示可能是分等级的，比如至少应该有俩个等级，一个是建议进行GC，一个是必须进行GC（再不GC就存不下了）。