armink/FlashDB #337: TSDB写入次数越多，写入时间越来越长

HenleyEn, 2024-12-18:
天龙哥，我需求是在1min记录一次系统运行的数据，一次记录大小小于64bytes，然后随着记录次数越来越多，需要写入时长越来越长，然后我做了一个简单的测试，每500ms插入一条日志：
在正常初始化后的插入数据非常快，但是后面每插入一条日志就会增加1ms
[image]
这是后续增加到一定程度：
[image]

运行过程：
```c
uint8_t log_tsl_append(fdb_tsdb_t tsdb, void *arg, int size)
{
    struct fdb_blob blob;
    uint8_t ret;

    ret = fdb_tsl_append(tsdb, fdb_blob_make(&blob, arg, size));
    if (ret == FDB_NO_ERR)
    {
        return LOG_SUCCESS;
    }
    return LOG_FAIL;
}

static void log_normal_append(fdb_tsdb_t tsdb)
{
    log_tsl_append(tsdb, &GasState, sizeof(GasState)) ;
    set_tsl_status(tsdb, tsdb);
}

void log_normal_append_data(void)
{
    TickType_t tick0 = xTaskGetTickCount();

    log_normal_append(&tsdb);

    TickType_t tick1 = xTaskGetTickCount();
    TickType_t tick = tick1 - tick0;
    log_i("write data spend times = %ld.%.3ld", (tick / configTICK_RATE_HZ), tick % configTICK_RATE_HZ);

    uint32_t last_time = 0;
    fdb_tsdb_control(&tsdb, FDB_TSDB_CTRL_GET_LAST_TIME, (void *)&last_time);
    log_i("last_time = %d", last_time);
}
```

armink, 2024-12-18:
有点奇怪，代码里是不存在线性增加的耗时动作的，你可以再深入排查在代码，看看是具体是哪块代码耗时导致的吗？

HenleyEn, 2024-12-18:
天龙哥，找到出现延长时间的地方了。
在这里，这个设置状态那里，调用了迭代器进行设置所有状态
```c
static void log_normal_append(fdb_tsdb_t tsdb)
{
    /* ...  */
    TickType_t tick_status_0 = xTaskGetTickCount();

    set_tsl_status(tsdb, tsdb);

    TickType_t tick_status_1 = xTaskGetTickCount();
    TickType_t tick_status = tick_status_1 - tick_status_0;
    log_i("log_normal_append : set status spend times = %ld.%.3ld", (tick_status / configTICK_RATE_HZ), tick_status % configTICK_RATE_HZ);
}
```
这个**set_tsl_status**调用：
```c
void set_tsl_status(fdb_tsdb_t tsdb, void *arg)
{
    { /* SET the TSL status */
        /* Change the TSL status by iterator or time iterator
         * set_status_cb: the change operation will in this callback
         *
         * NOTE: The actions to modify the state must be in orderC.
         *       like: FDB_TSL_WRITE -> FDB_TSL_USER_STATUS1 -> FDB_TSL_DELETED -> FDB_TSL_USER_STATUS2
         *       The intermediate states can also be ignored.
         *       such as: FDB_TSL_WRITE -> FDB_TSL_DELETED
         */
        fdb_tsl_iter(tsdb, set_status_cb, arg);
    }
}
```
在增加日志时，都会遍历一次。我移植的时候应该忽略了这一节：
[image]

现象：
[image]

修改后：
[image]
插入时长趋于稳定在20ms。

**我在网络上查看到有些遇到TSDB写入时间越来越长的问题，感觉应该也是这个问题，在对数据设置状态时没有修改你提供的示例，直接移植过来能用了，数据量大了才发现这个问题。**

armink, 2024-12-18:
如果是乌龙，那就把议题关闭吧

closed 2024-12-18 by HenleyEn