armink/FlashDB #340: 同名key不断的写入不同的内容，写满后不会会覆盖旧的，最终导致报gc fail，无法继续再写入

thisisluckywang, 2025-01-07:
I[00226783][0044]:[NA][*]test_storage_manager_read_db():buf:11:22:33:44:55:66:77:88-25878, ret: 0
E[00226783][0044]:[NA][*]test_storage_manager_read_db():crc16 cmp fail!, need: 0x0, but: 0x2597
D[00226784][0044]:[NA][storage_internal.c]kv_put_internal():======>put=START=>0,key:key_bt_mac,val_size:2048
D[00226784][0044]:[NA][*]alloc_kv():Trigger a GC check after alloc KV failed.

D[00226785][0044]:[NA][*]new_kv():Alloc an KV (size 2082) failed when new KV. Now will GC then retry.

D[00226785][0044]:[NA][*]gc_collect_by_free_C start-1I[00226794][0044]:[NA][*]do_gc():GC start-2222

D[00226828][0044]:[NA][*]do_gc():KV (10,key_bt_mac) is garbage NOT need move, collect it.

D[00226859][0044]:[NA][*]do_gc():Collect a sector @0x00004000

I[00226860][0044]:[NA][*]gc_collect_by_free_size():collect exit-333

D[00226962][0044]:[NA][*]alloc_kv():Trigger a GC check after alloc KV failed.

I[00226962][0044]:[NA][*]new_kv():Err: Alloc an KV (size 2082) failed after GC. KV full.

tag:kv_db_dump_internal,fdb_kvkey_bt_mac=blob @0x00006032/2048bytes
system_boot_count=blob @0x0000685b/4bytes

mode: next generation
size: 14821/28672 bytes,using(51%)

thisisluckywang, 2025-01-07:
最新的2.1.1版本，功能貌似很强，不过bug有点多啊

anujdeshpande, 2025-02-25:
@thisisluckywang how did you get around this issue?
@armink I am seeing something similar - can you confirm if this is a known bug?