armink/FlashDB #349: 当分区数据写满后，触发gc回收flash空间时，此时按复位重启，重启后，程序会卡到触发gc回收flash空间的过程中

qulei123, 2025-02-18:
1. 环境：

stm32F103 512K内部flash
裸机程序测试

2. 问题：
当分区数据写满后，触发gc回收flash空间时，此时按复位重启，重启后，程序会卡到触发gc回收flash空间的过程中flash全部擦除后，再烧写程序可以恢复正常

3. 日志如下：

```
nr@sky# I:tim clk 100khz, timeout 11800
[D/FAL] (fal_flash_init:47) Flash device |             stm32_onchip | addr: 0x08000000 | len: 0x00040000 | blk_size: 0x00000800 |initialized finish.
[I/FAL] ==================== FAL partition table ====================
[I/FAL] | name      | flash_dev    |   offset   |    length  |
[I/FAL] -------------------------------------------------------------
[I/FAL] | fdb_tsdb1 | stm32_onchip | 0x0001a000 | 0x00002000 |
[I/FAL] | fdb_kvdb1 | stm32_onchip | 0x0001c000 | 0x00004000 |
[I/FAL] =============================================================
[I/FAL] Flash Abstraction Layer (V0.5.99) initialize success.
[FlashDB][kv][env][fdb_kvdb1] (..\Modules\flashDB\src\fdb_kvdb.c:1802) The oldest addr is @0x00000000
[FlashDB][kv][env][fdb_kvdb1] (..\Modules\flashDB\src\fdb_kvdb.c:1818) KVDB size is 16384 bytes.
[FlashDB] FlashDB V2.1.1 is initialize success.
[FlashDB] You can get the latest version on https://github.com/armink/FlashDB .
==================== kvdb_basic_sample ====================
[FlashDB][kv][env][fdb_kvdb1] KV: boot_count - 0x00003654 - 0x00003688
get the 'boot_count' value is 52
set the 'boot_count' value to 53
===========================================================
==================== kvdb_type_string_sample ====================
[FlashDB][kv][env][fdb_kvdb1] (..\Modules\flashDB\src\fdb_kvdb.c:930) Trigger a GC check after alloc KV failed.
[FlashDB][kv][env][fdb_kvdb1] (..\Modules\flashDB\src\fdb_kvdb.c:1076) Alloc an KV (size 48) failed when new KV. Now will GC then retry.
[FlashDB][kv][env][fdb_kvdb1] (..\Modules\flashDB\src\fdb_kvdb.c:1162) The remain empty sector is 1, GC threshold is 1.
[FlashDB][kv][env][fdb_kvdb1] (..\Modules\flashDB\src\fdb_kvdb.c:1059) Moved the KV (username) from 0x00000024 to 0x00003824.
[FlashDB][kv][env][fdb_kvdb1] (..\Modules\flashDB\src\fdb_kvdb.c:1059) Moved the KV (password) from 0x0000005C to 0x0000385C.
[FlashDB][kv][env][f _   _ ____    __  __ _                  ____  _          _ _ 
| \ | |  _ \  |  \/  (_) ___ _ __ ___   / ___|| |__   ___| | |
|  \| | |_) | | |\/| | |/ __| '__/ _ \  \___ \| '_ \ / _ \ | |
| |\  |  _ <  | |  | | | (__| | | (_) |  ___) | | | |  __/ | |
|_| \_|_| \_\ |_|  |_|_|\___|_|  \___/  |____/|_| |_|\___|_|_|
                                                              
nr@sky# I:tim clk 100khz, timeout 11800
[D/FAL] (fal_flash_init:47) Flash device |             stm32_onchip | addr: 0x08000000 | len: 0x00040000 | blk_size: 0x00000800 |initialized finish.
[I/FAL] ==================== FAL partition table ====================
[I/FAL] | name      | flash_dev    |   offset   |    length  |
[I/FAL] -------------------------------------------------------------
[I/FAL] | fdb_tsdb1 | stm32_onchip | 0x0001a000 | 0x00002000 |
[I/FAL] | fdb_kvdb1 | stm32_onchip | 0x0001c000 | 0x00004000 |
[I/FAL] =============================================================
[I/FAL] Flash Abstraction Layer (V0.5.99) initialize success.
[FlashDB][kv][env][fdb_kvdb1] (..\Modules\flashDB\src\fdb_kvdb.c:1802) The oldest addr is @0x00000000
[FlashDB][kv][env][fdb_kvdb1] (..\Modules\flashDB\src\fdb_kvdb.c:1818) KVDB size is 16384 bytes.
[FlashDB][kv][env][fdb_kvdb1] (..\Modules\flashDB\src\fdb_kvdb.c:1162) The remain empty sector is 0, GC threshold is 1.
[FlashDB][kv][env][fdb_kvdb1] (..\Modules\flashDB\src\fdb_kvdb.c:1131) KV (username) is garbage NOT need move, collect it.
[FlashDB][kv][env][fdb_kvdb1] (..\Modules\flashDB\src\fdb_kvdb.c:1131) KV (password) is garbage NOT need move, collect it.
[FlashDB][kv][env][fdb_kvdb1] (..\Modules\flashDB\src\fdb_kvdb.c:1131) KV (boot_count) is garbage NOT need move, collect it.
 _   _ ____    __  __ _                  ____  _          _ _ 
| \ | |  _ \  |  \/  (_) ___ _ __ ___   / ___|| |__   ___| | |
|  \| | |_) | | |\/| | |/ __| '__/ _ \  \___ \| '_ \ / _ \ | |
| |\  |  _ <  | |  | | | (__| | | (_) |  ___) | | | |  __/ | |
|_| \_|_| \_\ |_|  |_|_|\___|_|  \___/  |____/|_| |_|\___|_|_|
                                                              
nr@sky# I:tim clk 100khz, timeout 11800
[D/FAL] (fal_flash_init:47) Flash device |             stm32_onchip | addr: 0x08000000 | len: 0x00040000 | blk_size: 0x00000800 |initialized finish.
[I/FAL] ==================== FAL partition table ====================
[I/FAL] | name      | flash_dev    |   offset   |    length  |
[I/FAL] -------------------------------------------------------------
[I/FAL] | fdb_tsdb1 | stm32_onchip | 0x0001a000 | 0x00002000 |
[I/FAL] | fdb_kvdb1 | stm32_onchip | 0x0001c000 | 0x00004000 |
[I/FAL] =============================================================
[I/FAL] Flash Abstraction Layer (V0.5.99) initialize success.
[FlashDB][kv][env][fdb_kvdb1] (..\Modules\flashDB\src\fdb_kvdb.c:1802) The oldest addr is @0x00000000
[FlashDB][kv][env][fdb_kvdb1] (..\Modules\flashDB\src\fdb_kvdb.c:1818) KVDB size is 16384 bytes.
[FlashDB][kv][env][fdb_kvdb1] (..\Modules\flashDB\src\fdb_kvdb.c:1162) The remain empty sector is 0, GC threshold is 1.
[FlashDB][kv][env][fdb_kvdb1] (..\Modules\flashDB\src\fdb_kvdb.c:1131) KV (username) is garbage NOT need move, collect it.
[FlashDB][kv][env][fdb_kvdb1] (..\Modules\flashDB\src\fdb_kvdb.c:1131) KV (password) is garbage NOT need move, collect it.
[FlashDB][kv][env][fdb_kvdb1] (..\Modules\flashDB\src\fdb_kvdb.c:1131) KV (boot_count) is garbage NOT need move, collect it.
 _   _ ____    __  __ _                  ____  _          _ _ 
| \ | |  _ \  |  \/  (_) ___ _ __ ___   / ___|| |__   ___| | |
|  \| | |_) | | |\/| | |/ __| '__/ _ \  \___ \| '_ \ / _ \ | |
| |\  |  _ <  | |  | | | (__| | | (_) |  ___) | | | |  __/ | |
|_| \_|_| \_\ |_|  |_|_|\___|_|  \___/  |____/|_| |_|\___|_|_|
                                                              
nr@sky# I:tim clk 100khz, timeout 11800
[D/FAL] (fal_flash_init:47) Flash device |             stm32_onchip | addr: 0x08000000 | len: 0x00040000 | blk_size: 0x00000800 |initialized finish.
[I/FAL] ==================== FAL partition table ====================
[I/FAL] | name      | flash_dev    |   offset   |    length  |
[I/FAL] -------------------------------------------------------------
[I/FAL] | fdb_tsdb1 | stm32_onchip | 0x0001a000 | 0x00002000 |
[I/FAL] | fdb_kvdb1 | stm32_onchip | 0x0001c000 | 0x00004000 |
[I/FAL] =============================================================
[I/FAL] Flash Abstraction Layer (V0.5.99) initialize success.
[FlashDB][kv][env][fdb_kvdb1] (..\Modules\flashDB\src\fdb_kvdb.c:1802) The oldest addr is @0x00000000
[FlashDB][kv][env][fdb_kvdb1] (..\Modules\flashDB\src\fdb_kvdb.c:1818) KVDB size is 16384 bytes.
[FlashDB][kv][env][fdb_kvdb1] (..\Modules\flashDB\src\fdb_kvdb.c:1162) The remain empty sector is 0, GC threshold is 1.
[FlashDB][kv][env][fdb_kvdb1] (..\Modules\flashDB\src\fdb_kvdb.c:1131) KV (username) is garbage NOT need move, collect it.
[FlashDB][kv][env][fdb_kvdb1] (..\Modules\flashDB\src\fdb_kvdb.c:1131) KV (password) is garbage NOT need move, collect it.
[FlashDB][kv][env][fdb_kvdb1] (..\Modules\flashDB\src\fdb_kvdb.c:1131) KV (boot_count) is garbage NOT need move, collect it.
 _   _ ____    __  __ _                  ____  _          _ _ 
| \ | |  _ \  |  \/  (_) ___ _ __ ___   / ___|| |__   ___| | |
|  \| | |_) | | |\/| | |/ __| '__/ _ \  \___ \| '_ \ / _ \ | |
| |\  |  _ <  | |  | | | (__| | | (_) |  ___) | | | |  __/ | |
|_| \_|_| \_\ |_|  |_|_|\___|_|  \___/  |____/|_| |_|\___|_|_|
                                                              
nr@sky# I:tim clk 100khz, timeout 11800
[D/FAL] (fal_flash_init:47) Flash device |             stm32_onchip | addr: 0x08000000 | len: 0x00040000 | blk_size: 0x00000800 |initialized finish.
[I/FAL] ==================== FAL partition table ====================
[I/FAL] | name      | flash_dev    |   offset   |    length  |
[I/FAL] -------------------------------------------------------------
[I/FAL] | fdb_tsdb1 | stm32_onchip | 0x0001a000 | 0x00002000 |
[I/FAL] | fdb_kvdb1 | stm32_onchip | 0x0001c000 | 0x00004000 |
[I/FAL] =============================================================
[I/FAL] Flash Abstraction Layer (V0.5.99) initialize success.
[FlashDB][kv][env][fdb_kvdb1] (..\Modules\flashDB\src\fdb_kvdb.c:1802) The oldest addr is @0x00000000
[FlashDB][kv][env][fdb_kvdb1] (..\Modules\flashDB\src\fdb_kvdb.c:1818) KVDB size is 16384 bytes.
[FlashDB][kv][env][fdb_kvdb1] (..\Modules\flashDB\src\fdb_kvdb.c:1162) The remain empty sector is 0, GC threshold is 1.
[FlashDB][kv][env][fdb_kvdb1] (..\Modules\flashDB\src\fdb_kvdb.c:1131) KV (username) is garbage NOT need move, collect it.
[FlashDB][kv][env][fdb_kvdb1] (..\Modules\flashDB\src\fdb_kvdb.c:1131) KV (password) is garbage NOT need move, collect it.
[FlashDB][kv][env][fdb_kvdb1] (..\Modules\flashDB\src\fdb_kvdb.c:1131) KV (boot_count) is garbage NOT need move, collect 
```

zhenbin-lin, 2025-04-08:
我也遇到这个问题了

zhenbin-lin, 2025-04-08:
> 我也遇到这个问题了

我解决了，发现是Flash的擦除问题，没擦除成功，导致一直触发GC

qulei123, 2025-04-09:
> > 我也遇到这个问题了
> 
> 我解决了，发现是Flash的擦除问题，没擦除成功，导致一直触发GC

怎么解决的，修改了哪些文件，我也试试，谢谢！

zhenbin-lin, 2025-04-09:
> > > 我也遇到这个问题了
> > 
> > 
> > 我解决了，发现是Flash的擦除问题，没擦除成功，导致一直触发GC
> 
> 怎么解决的，修改了哪些文件，我也试试，谢谢！

你先确保 FAL 对你Flash的擦除是成功，我遇到的问题是我擦除 Flash 失败，但是我返回了擦除成功的结果的 GC，导致出现了 GC 一直触发的问题。你最好检查一下你自己实现的 FAL 的 flash 操作是不是都正确的。